CTCaer/hekate #426: Launching Atmosphere creates Cyan Screen

Urbanxx001, 2020-05-09:
I'm not sure on which end this is an issue (hekate, atmosphere, or user error), but when I launch CFW for Emu, first it flashes the hekate logo and then an unresponsive bright cyan screen. Thankfully I can exit it by holding the power button, but I'm lost as to resolving it. I've tried the following:

Updated both Hekate and Atmosphere, on SD card and payload injector
Recreated Emu on SD partition
Removed 'kip1patch=nosigchk' and 'kip1=atmosphere/kips/*' lines in hekate_ipl.ini
Removed CFW (Edizon, Goldleaf, etc.)

CTCaer, 2020-05-09:
Did you updated your emummc with choinx?
If yes, check that your pkg1 and pkg2 match.

Urbanxx001, 2020-05-09:
I don't use ChoiNX, only updated using the SD card, RCMloader brand injector, and (initially) Tegra.

CTCaer, 2020-05-09:
Normally issues with blue screen are because of mismatches or corruption.

Can you get a photo or screenshot of Console info -> Sd card?

Urbanxx001, 2020-05-09:
Here's the SD info: [https://imgur.com/a/H3OyofV](url)

CTCaer, 2020-05-10:
Sd card seems legit. Only issue is that it says A2. And you can't have A2 on a U1/V10 card.
(Also sandisk doesn't make such cards.)

Can you take a macro photo of the sd card with good lighting?
Also try the h2testw or another sd card checker.

Urbanxx001, 2020-05-10:
After inspecting, you're right the info says A2 even though the card is A1. Here it is: https://imgur.com/a/LmatNVN

I don't know if the discrepancy was before or after the update, but I was able to use EmuNand before. I'll run the h2testw and report back. Thank you so much for your help

CTCaer, 2020-05-10:
In case the test is good:
Have in mind that normally a blue screen happens when there's pkg2 mismatch or sth went wrong with the update.
Maybe you can try to manually re-update to 10.

closed 2020-06-01 by CTCaer