CTCaer/hekate #428: File-based emunans stuck on NS bootlogo

rashevskyv, 2020-05-09:
Tried to make Emu with SX and with hekate - same result. Tried exFAT and FAT32. 
10.0.2, ATM 0.12.0

CTCaer, 2020-05-09:
How you created it?
File based or partition based?

Show me a photo or screenshot from Nyx **0.9.1** > Console info > Sd card

rashevskyv, 2020-05-09:
File-based. 
Partition-based works without issues

rashevskyv, 2020-05-09:
I tried different cards

Samsung EVO 64 FAT32:
[image]

SanDisk Ultra 128 exFAT:
[image]

Viperbite, 2020-05-09:
[image]
Similar problem, file based emunand stops after the Nintendo logo. Created in the CX OS or via Hecate. Tested for FAT32 and exFAT.

CTCaer, 2020-05-09:
All cards are legit.
How you boot it? I need the boot entry used in the .ini

rashevskyv, 2020-05-09:
```
[config]
autoboot=1
autoboot_list=0
bootwait=3
customlogo=1
verification=2
backlight=40
autohosoff=0
logopath=bootloader/bootlogo.bmp
autonogc=0
{ }
[Atmosphere]
payload=bootloader/payloads/fusee-primary.bin
{ 
```}

Partition-based boots fine with this ini

Cards are legit, but I can test it with h2testw if you need

CTCaer, 2020-05-09:
Well, there's your issue. emuMMC code is something that bootloader injects.
Each bootloader has its own method.
For now the fast file-based can only be attained by using hekate to boot atmosphere (fss0 or the old kip style).
If you still want to use fusee-prim to boot, just wait some minutes and it will boot. Or wait until it gets updated.

Also your config needs cleaning. Go to options tab (last tab) once and press save options to fix it.

closed 2020-05-09 by CTCaer