CTCaer/hekate #429: beckaup emmc

kristal91, 2020-05-09:
hi is it possible to create a beckaup from an emmc that I have on the sd and restore it on the original nand? I broke the nand and I don't have the beckaup automatically so it would be beckaup from the emmc I have and restore it

CTCaer, 2020-05-09:
Yes you can. What type of emuMMC you have?

kristal91, 2020-05-09:
i have emmc created by hekate

CTCaer, 2020-05-09:
file based or partition based?

kristal91, 2020-05-09:
partition based

CTCaer, 2020-05-09:
Go to backup and press the button on the bottom. The one that enables SD raw based partition.
The buttons will change names from eMMC to emuMMC.
Backup both boot0/1 and raw GPP.

kristal91, 2020-05-09:
after the beckaupnsd emummc raw reset the state from sd to nand and restore the beckaup on the nand?

CTCaer, 2020-05-09:
you need to move the files into the restore folder

kristal91, 2020-05-09:
okok now i feel thank you

CTCaer, 2020-05-10:
You probably already finished your restoration project.
Closing issue.

closed 2020-05-10 by CTCaer

kristal91, 2020-05-10:
yes thanks