CTCaer/hekate #430: Checksums do not mutch when mount BOOT0\1 with hekate

rashevskyv, 2020-05-09:
Trying to write Boot0\1 with etcher, but got an error: 

[image]

With memloader_v3 all going well

CTCaer, 2020-05-09:
Which etcher version?

rashevskyv, 2020-05-09:
It was 1.5.76
On the last version I got this - 

[image]

CTCaer, 2020-05-10:
Just found time to test and do wireshark dumps.

Issue is that it works splendidly here (balenEtcher v1.5.86):
[image]
Wireshark dumps are also perfect. No issues or retries.

Have in mind that hekate is more strict with cable issues. It can detect a disconnection immediately.
Check that your cable and its connectors are fine.
(Btw, I did the tests with a nintendo cable and another 2 which were usb2 and usb3, both with proper cable impedance.)

As for the strictness, I can try to ease it up in next versions.
For now, try to keep your cable steady and make sure that is nicely connected.

closed 2020-06-01 by CTCaer