CTCaer/hekate #431: [Enhancement] Backup eMMC and emuMMC via USB

NickHam13, 2020-05-10:
First things first, I'm sorry if this is already a feature and I've somehow missed it, or already a planned addition.

I think the option to backup your eMMC and emuMMC via USB would be a great quality of life feature. I don't have much space left on my SD card but would still like to make full backups directly to my PC without going through the trouble of joining all the files. 

And to end things off, thank you for your continued work on this project. I'm excited to see what you have in store for this year.

CTCaer, 2020-05-10:
Hi.
This is already supported.

Tools -> USB Tools -> Choose an eMMC or emuMMC button -> Use a tool to backup the mounted disk.

NickHam13, 2020-05-10:
Is there a specific tool you can recommend for this?

CTCaer, 2020-05-10:
There are many and depend on your OS

leandrodeassis, 2020-05-26:
> Is there a specific tool you can recommend for this?

I suggest NxNandManager

closed 2020-06-15 by CTCaer