CTCaer/hekate #434: Can't boot into cfw through hekate no matter what i do

ItsJoe98, 2020-05-14:
So i bought a sandisk 256 sd card from argos and i did everything the same as i did on my 128gb sd that works still, i can't clone it because it just says there are bad sectors so i had to start fresh but on 10.0.2 ofw it just doesn't work i get the pkg1 error on both sd cards when i try launch emummc but when i launch the fuseee bin payload the 129gb sd works and boots into cfw but the 256 freezes to black screen. i have been trying this all day with all kinds of solutions not working. i know you don't use this for support but will help out when you can so that would be appreciated.

CTCaer, 2020-05-14:
Your sd card reader has issues probably.
Go to Console info -> SD card and get me a photo or screenshot of BOTH cards.

Also have in mind that emummc.ini sector key should always match emuMMC's start on the disk

ItsJoe98, 2020-05-14:
nevermind i figured it out it was fake sd card 20 down the drain thanks anyway

closed 2020-05-14 by ItsJoe98