CTCaer/hekate #436: How to merge 4 sd cards (sysnand games, emummc, android, linux)

ImEvwan, 2020-05-15:
The title. If anything is there a way to at least keep my emummc because I bought a new SD card that was much larger but never had a way to combine everything

CTCaer, 2020-05-16:
Yeah you can but you need to make images of each partition, make partitions via hekate and then dd back the images. FAT32/exFAT partition can just be normally merged and backed up.
Also hekate can backup emummc if it was made via it. And after dd you can just fix the offset.

ImEvwan, 2020-05-20:
I'm sorry for my ignorance here, but what is DD? Is that a feature in a program? It may be a bit to ask, but can you give an answer that would point me in the right direction as I have no clue where to even look. Thanks!

CTCaer, 2020-05-20:
dd is a command in Linux (normally) where, in your case, it can copy disk data directly to files.
Basically what you need to do is to create partition images. And after creating the partitions in the sd card that will hold everything, you can copy them back.

ImEvwan, 2020-05-20:
ah, didnt even cross my mind that this would be a terminal command. So i'd disk dump the SD cards, partition up the SD card and write the images to each partition individually?

CTCaer, 2020-05-20:
Yeah. And it probably will just work. The only thing that needs a bit of change is emuMMC raw.
You will need to change emummc.ini and raw_based to point to where BOOT0 is. Normally emummc part + 16MiB.

closed 2020-06-01 by CTCaer