CTCaer/hekate #437: Verification failed. Help!

justacooldudeuknow28, 2020-05-19:
Hi,

Hi.

I wanted to backup my nand using hekate. This is before I install atmosphere. So I can have a clean nand backup (before homebrew).

So I am using TegraRcmGUI 2.6 and the version of hekate for switch ofw 10.0.2. I have tried 2, 64gb sd cards and it backups my Backup eMMC Boot 00 and Boot 01 successfully but then after it say's verification failed on backup eMMC Raw GPP, (00: rawnand.bin...). Is that because there is nothing (no files on my switch) Because the switch is clean?

Why would eMMC Raw GPP, (00: rawnand.bin...) fail verification?

Are there any other ways to backup my nand? So I can have a clean backup (before homebrew?). Thanks

Also,
Could it be the cards are defective?
I used 2, micro center sd 64gb cards.

I ordered a samsung 64gb evo card to try that is coming from Amazon tommorrow.

CTCaer, 2020-05-19:
Get me a photo or screenshot of Console info -> Sd card for both cards.

justacooldudeuknow28, 2020-05-19:
> Get me a photo or screenshot of Console info -> Sd card for both cards.

I am going to try the new sd card. If it works I'll let you know?

You wanted me to send a photo or screenshot of what? My switch consoles info? 

And then you wanted to also see my cards info too? Is that all correct?

justacooldudeuknow28, 2020-05-19:
Hi,

Got it working with the new sd card. Thanks anyhow. 👍

CTCaer, 2020-05-20:
Probably fake cards.

closed 2020-05-20 by CTCaer