CTCaer/hekate #440: Hekate stuck at logo screen, only works without Nyx.bin

tonyhoro, 2020-05-22:
This issue seems to be very specific, so after trying to get help from people at gbatemp and reddit, I'm trying to reach out to you guys.

This issue started when I tried to open Edizon, but it froze so I had to manually power off. After that, hekate text got all messed up. I managed to boot Atmosphere again with their primary payload, but edizon fucked up everything again, and now hekate gets stuck at the logo screen when launching the payload via tegraRCM. Atmosphere can still be booted up by loading its payload.

I'm on the latest atmosphere and hekate, OFW/CFW is on 10.0.2. I mentioned Edizon because when I talked about my issue on reddit, another guy had the same problem with this homebrew, which seems to be some useful clue and a way to reproduce the error.

Also, he told me that deleting nyx makes hekate boots, which worked.

https://www.reddit.com/r/SwitchPirates/comments/gonlwx/cant_boot_on_hekate_anymore_stuck_at_hekate_ipl/

CTCaer, 2020-05-22:
Use release zip to re-copy-paste the bootloader folder.
Your files are corrupted. So stop using exFAT.

And stop pirating >SwitchPirates

closed 2020-05-22 by CTCaer

tonyhoro, 2020-05-22:
> Use release zip to re-copy-paste the bootloader folder.
> Your files are corrupted. So stop using exFAT.
> 
> And stop pirating >SwitchPirates

1) I actually recopied all files from the bootloader folder but no use
2) I'm planning to stop using exFAT anyway, I thought that maybe this issue wasn't related with this, since everything else is running fine on CFW.
3) Please don't assume I'm pirating. I just posted it on all subreddits that had people who uses CFW.

deadair3210, 2020-05-22:
Likely the filesystem itself is damaged, so recopying the files wont do much. Ninty's exfat driver is hot garbage so try to switch to fat32 first

CTCaer, 2020-05-22:
Still avoid exFAT.
And what deadair said is true. If file allocation table is corrupted then recopying will not fix it.