CTCaer/hekate #443: Soft-Brick After Creating emuMMC

cDOTwav, 2020-05-31:
I'm having trouble with hekate and emuMMC. I had trouble with this in the beginning of me installing CFW and creating a emuMMC, in which it would just soft-brick when loading into Atmosphere, so I abandoned using hekate and just booted straight into Atmoshere. Flashforward to now and it's the same problem. It boots up fine when powering on to stock firmware, but when it comes to booting into CFW it just soft-bricks and I have to turn my Switch off. Any idea why this is happening? Maybe I'm missing a step?

CTCaer, 2020-05-31:
Black screen is not a soft-brick.
What type of emummc you created? Also take a photo or screenshot from Console Info -> Sd card

cDOTwav, 2020-05-31:
> Black screen is not a soft-brick.
> What type of emummc you created? Also take a photo or screenshot from Console Info -> Sd card

I saved it as an SD file 

[image]

CTCaer, 2020-05-31:
Your sd is legit.
Currently sd file based only works properly if booted natively via hekate.
In both cases you booted via fusee-prim (chainload via hekate also counts as not using hekate's native boot). You need to use the fss0 method.

closed 2020-05-31 by CTCaer