CTCaer/hekate #445: [Bug]  Not able to launch HOS 10.0.4

tk1076408, 2020-06-05:
Description:
System firmware: 10.0.4
Atmosphere: 0.12.0
Hekate: 5.2.1
SD card: Samsung EVO 64BG

A clean SD card, created emuMMC then launch shows error:
```
Failed to apply 'nosigchk'
Failed to launch HOS!
```

Let me know what else info that can help debug. Thanks!

m4xw, 2020-06-05:
Kek

deadair3210, 2020-06-05:
don't use sig patches and it will boot fine

closed 2020-06-05 by CTCaer

tk1076408, 2020-06-05:
> don't use sig patches and it will boot fine

With that config line removed, I am able to launch HOS, but after that, Goldleaf not able to install NSP from SD card. Reported issue here https://github.com/XorTroll/Goldleaf/issues/556
Do you think it’s related to this?

Thanks!

CTCaer, 2020-06-05:
No piracy discussion here