CTCaer/hekate #446: Failed to init eMMC when launching fss0 with emummc without an eMMC

dennthecafebabe, 2020-06-06:
I bricked my unit's eMMC today and after removing it I noticed that even though the boot entry I'm launching looks like this:
```ini
[emu]
fss0=atmosphere/fusee-secondary.bin
emummcforce=1
``` 
(also tested with `atmosphere=1` on there at some point)
and a working emummc setup:
```ini
[emummc]
enabled=1
sector=0x13469000
path=emuMMC/RAW1
id=0x0000
nintendo_path=emuMMC/RAW1/Nintendo
```
I get the following output:
```
Initializing...
Failed to init eMMC
Failed to init HOS.
```

I'm sorry if this is intentional or if I'm missing something in my config, just thought this might maybe be unwanted behaviour since I personally wouldn't have expected to need a working emmc to boot emummc.

CTCaer, 2020-06-07:
>I personally wouldn't have expected to need a working emmc to boot emummc

And you are right. Currently depends on it. But the check is there for one reason. emuMMC doesn't work without it. The emuMMC driver expects that normal ops are working, so they don't need to be emulated. A fully mmc emulation needs a huge amount of time in order to do it.

Is your eMMC toasted or just corrupt?
If corrupt, you can just restore a backup or just put it back if only corrupted. In both cases here it will work.
If toasted or removed, it will just not work.

dennthecafebabe, 2020-06-07:
Ah, didn't knew that - thanks for letting me know!

> Is your eMMC toasted or just corrupt?

Toasted comes closer I suppose but it isn't that big of a deal since I had a couple spare ones laying around anyway and my console is back on working again - just figured I'd open an issue since I was curious about this.

This can probably be closed then, my bad.

CTCaer, 2020-06-08:
Well using a whatever off the shelve emmc just makes it work yeah.

But as I described, the current limitation is not on a bootloader level.
The error message is forced in the bootloader in order to not just get greeted with a classic black screen hang in HOS.

closed 2020-06-08 by CTCaer