CTCaer/hekate #447: Wrong Battery percentage please help

cowkiller04, 2020-06-08:
OKay here is what's going on:

Updated to latest atmosphere, CFW and Hekate. All of a sudden I noticed that battery drain is really really slow... after playing for two hours battery still in 80%. Then once I hit 49% it will tell me battery level low (goes to sleep with the red icon) system eventually shuts down. Okay so I figured that Switch is confused and that 50% is now the 1% (pls note that the length of playing time is normal, meaning battery is not the problem rather it is the indicator that is messed up). 

I charge the switch, normal charging hours of 3 hours. After 3 hours it is at 50% (cause 50% is the new 100%). So I said okay that is fine as normal playing time is still 3 to 4 hours +. I played for like two hours before I slept and battery level says 35%. I turned off the switch. Next day I open switch and battery is at 100%???? wtf?

Basically, the % of the battery is all messed up. Is there a way to fix this like a reset/resync/recalibrate battery option?

Thank you

CTCaer, 2020-06-08:
There's no way to fix that ouside of HOS.
For now you will need to charge and discharge inside HOS, until it gets fixed on its own.
The percentages mean nothing with that issue. You can easily see that by the voltage levels.
3200mV is completely empty and 4200mV is full.

cowkiller04, 2020-06-08:
@CTCaer thank you for the reply, what is the meaning of HOS? Sorry noob.

cowkiller04, 2020-06-08:
This is my true current batter level now right? Do you know what is causing this? 

[image]

masagrator, 2020-06-08:
HOS means Horizon OS which is native Nintendo Switch firmware where you play Nintendo Switch games.

cowkiller04, 2020-06-08:
okay this means I have to use the switch in HOS until battery recalibrate

cowkiller04, 2020-06-08:
@masagrator heard there was a "dsync battery" option in Hekate before?

deadair3210, 2020-06-08:
That was for something else and the issue is automatically fixed when present now. The only way to fix your issue is to simply run the battery down while in the normal firmware and the firmware will slowly recalibrate to the actual amount of charge

CTCaer, 2020-06-15:
As said before, you will need to charge and discharge inside HOS until it fixes itself for now

closed 2020-06-15 by CTCaer