CTCaer/hekate #449: Request for tutorial on partitioning with hekate

Expertsonicgamer32x, 2020-06-14:
This is one of the most brilliant additions I've ever seen In hekate and I don't have a Linux computer and I want to know if there's a tutorial on how to do this iirc you don't need a Linux computer to do this right?

CTCaer, 2020-06-14:
Well if you can get the correct android (pie) and linux images you don't need a windows pc.
Also dunno what you want on that tutorial.

In hekate you just move sliders and repartition. And that's it. After that you can flash a linux partition image or twrp so you can install android.

Expertsonicgamer32x, 2020-06-15:
Welli still don't understand ''pie'' but what I want to do is use Android and emummc on the same sd card

Expertsonicgamer32x, 2020-06-15:
And what's the file path for emummc and andriod?

CTCaer, 2020-06-15:
You don't need to worry for that for now. When Linux and Android release official support, you will only need to drag n drop a folder and tap some buttons.
For now you need to build Android pie on your own.

As for emuMMC, you can just create one from inside hekate.

Expertsonicgamer32x, 2020-06-16:
Ok thanks

closed 2020-06-16 by Expertsonicgamer32x