CTCaer/hekate #450: Joy Cons Not Working on v5.3.0

lowercasejamie, 2020-06-15:
Hi all,

Having a slight issue with my Joy Cons not working when in Hekate v5.3.0.

My touch screen does not recieve touch, so I am left to use my Joy Cons in order to navigate the home screen. When greeted with the "Enter Date and Time for Nyx" screen, I am unable to do so. I have seen that other users have a white dot which they can use the analogue stick to navigate with, but I don't have one.

The Joy Cons vibrate when connecting to the console and the screen is not in a frozen state (unplugging a charger changes from "charging" icon to normal battery icon).

Hope that explains my issue clearly enough, thanks in advance for any and all responses.

CTCaer, 2020-06-15:
You should be able to use the joy con in that screen.
Maybe you pressed the stick early and calibration (happens when they vibrate) managed to stuck the cursor.

You can reload or just pull the left joycon out and replug it. This will recalibrate them

lowercasejamie, 2020-06-16:
Thank you for the quick response.

I will try that tonight when I get home and get back to you.

lowercasejamie, 2020-06-16:
> You should be able to use the joy con in that screen.
> Maybe you pressed the stick early and calibration (happens when they vibrate) managed to stuck the cursor.
> 
> You can reload Nyx or just pull the left joycon out and replug it. This will recalibrate them.
> 
> Edit:
> Also if your touchscreen is not broken, you could try using the tuning calibration tool.

Hi there,

I've tried disconnecting and reconnecting the joy cons and I am met with the same issue. There is no cursor present on the screen.

CTCaer, 2020-06-16:
Can you try this: go to bootloader/sys/ and rename nyx.bin to nyx.bak and throw there the nyx.bin from the v5.2.1 release. Tell me what happens.

Note: Under no circumstance should joycon stop working. You can't actual feel the vibration if the initialization stage of them doesn't finish.

lowercasejamie, 2020-06-16:
I get greeted with the screen seen here -
[image]

Then, after a short while, the screen shown here appears. However, I cannot do anything as the touch function no longer works and my Joy Cons are not registering

CTCaer, 2020-06-16:
So it's not a thing that was changed in new nyx. I wonder why your joycon do not work.
Please find me in retronx discord so we can do some tests

lowercasejamie, 2020-06-16:
Okay no worries - can you post a link for the discord please?

CTCaer, 2020-06-16:
So the issue here is that in this particular unit, the touch chip seems dead.
That forces the i2c bus to timeout. The timeout takes 1500ms.
This makes the joycon to timeout because they expect a communication for at least every 600ms.

The following commit fixes that, by actually checking if init succeeded:
https://github.com/CTCaer/hekate/commit/3b0925b912246a8b1c0dc2ecc62e09a5c9ae7788

closed 2020-06-16 by CTCaer