CTCaer/hekate #451: Will not boot beyond the Hekate part of the chainloading process with Atmosphere 0.13.0

dothackjhe, 2020-06-16:
Hekate 5.3.0 struggles to boot past the Hekate step in the chainloading process when using the files for Atmosphere 0.13.0.

Conversely, the system boots just fine with Atmosphere 0.12.0.

CTCaer, 2020-06-16:
What do you mean precisely? What are the steps to reproduce that?

dothackjhe, 2020-06-16:
I think this may be a false alarm. I have not have the time to test everything yet, but I think the fault lies in how I do the supposed chainloading. 

Based on another person's comment on GBATemp, I might be booting the game via a modified Atmosphere (CFW SYSNAND) by default, instead of fusee-primary.bin.

CTCaer, 2020-06-16:
You have a wrong notion of what is chainloading.
Also your message implies piracy.

closed 2020-06-16 by CTCaer

dothackjhe, 2020-06-16:
So, I just tested the hypothesis provided to me. Apparently, there is indeed an issue with Hekate 5.3.0 having trouble chainloading fusee.primary.bin for Atmosphere 0.13.0.

CTCaer, 2020-06-16:
Well the issue is not hekate. And that's why I said you have a wrong notion of chainloading.
You obviously have mixed files and expect 0.13 to run with 0.12 parts.
Also if your Atmosphere (CFW SYSNAND) was using hekate to boot, it would have just worked.

dothackjhe, 2020-06-16:
I tried both ways: it did not work.