CTCaer/hekate #452: Unable to compile hekate payload - build error

raytan1807, 2020-06-16:
[image]
Hi, not sure if anyone facing this error when trying to build the file but previously I have no encounter any error when building the hekate v5.2.1 & Nyx v0.9.1. I run cmd as administrator and put in the "make" command and execute makefile in the source directory. Sorry, not sure whether is it appropriate to post and ask here but appreciate if someone could help. Thanks.

CTCaer, 2020-06-16:
What's that? msys?
(Because it works on linux and wsl1)

raytan1807, 2020-06-16:
msys2, I set up the building environment using devkitpro 3.0.3. does it support that ? as I was able to just build it without any error for the last version. thanks for your quick reply :)

CTCaer, 2020-06-16:
Fixed with https://github.com/CTCaer/hekate/commit/c5f032fcb2cd6eba9899249387ee736ef1ebdcf8

closed 2020-06-16 by CTCaer

raytan1807, 2020-06-16:
Thanks for the quick fix !! I can build without issues now !! :D