CTCaer/hekate #454: [Feature Request] Add option to use animated bootlogo

MudkipBanana, 2020-06-16:
As shown in this proof of concept here: https://twitter.com/CTCaer/status/1123279256660992000

masagrator, 2020-06-16:
Well, this is not animated logo, this are raw images shown one by one. And each one takes few MB (if it's RGBA8, it's 3.5 MB), so for 30 frames you are getting 105 MB.

MudkipBanana, 2020-06-16:
Oh, ok. I don't really know that much about how the switch works, but given enough space, could you still do it? I have a 400gb micro sd card.

masagrator, 2020-06-16:
Well, for Atmosphere you need to edit the part where it shows logo and instead shows each image in some time intervals. And you need to compile it, it will not load logo from external image as it is.

Idk if this can be bypassed by Hekate (i mean loading external images)

MudkipBanana, 2020-06-16:
How would one go about doing this?

CTCaer, 2020-06-16:
That's hekate @masagrator 
The cfw has nothing to do with that.

This will be released when I finalize the format of that and add support for sound actually.
For now it remains a PoC that can reach 60fps.

But it's not high on my todo list. Mainly because any reasonable animation, basically raises boot times.
I'll close this, but have in mind that I always think of it from time to time.
We'll see in the future.

closed 2020-06-16 by CTCaer