CTCaer/hekate #456: Hekate doesn't boot...

ryancruisex, 2020-06-18:
Already viewed the previous issue about this.

Using fusee launcher OSX.

Mine is different;

I sent hekate payload, atmosphere splash loads up then nintendo logo and i can then access home-brew menu through photos. (hack successful)

I then installed hekate through home-brew menu however it doesn't appear in tools, only in installed section and only gives me an option to reinstall or delete. i'm wanting to open the hekate menu where it lets me create a emuMMC. how do I access it?

CTCaer, 2020-06-18:
hekate is not a homebrew app for HOS (Switch OS) that you can launch like that.
It's a payload that you inject or reboot into. 

Download the latest from the releases tab here and drag n drop the bootloader folder from inside the zip into your sd card root.
If a message asks you to merger and replace files, click yes.

Then inject the hekate v5.3.0 .bin.
Because you probably enabled autoboot without knowing, press VOL- while injecting so you can go to hekate menu.

closed 2020-06-18 by CTCaer

ryancruisex, 2020-06-18:
i just did this and for the first time got the text on the screen, however now its saying 'could not open /bootloader/hekate_ipl.ini'
make sure it exists

CTCaer, 2020-06-18:
You need a `hekate_ipl.ini` if you want to boot CFW or sth else.

A very simple one that boot Atmosphere is:

```ini
[config]
autoboot=0
autoboot_list=0
bootwait=3
backlight=100
autohosoff=0
autonogc=1
updater2p=1

[Atmosphere SYS]
fss0=atmosphere/fusee-secondary.bin
emummc_force_disable=1
{}

[Atmosphere EMU]
fss0=atmosphere/fusee-secondary.bin
emummcforce=1
{}

```

First entry boots sysnand and second boots emuMMC.

ryancruisex, 2020-06-18:
nevermind, merged the new file with old and now hekate boots up after injection w vol - held, thank you