CTCaer/hekate #458: Save file issue

I8ikura, 2020-06-18:
I tried to load atmosphere on my switch today to edit my Pokemon Sheild save file and got an error that said "unkown pkg1 version". So I looked up how I could fix this and the guide (https://github.com/CTCaer/hekate/issues/347) told me to click emuMMC migration. I did and this time atmosphere was able to load. However when I did this my switch went back to how it was when I first loaded CFW on it. My pokemon save file was also reverted back and I can seem to bring back to current save file. Any way to make it so my current switch menu is displayed instead of one from 6 months ago?

CTCaer, 2020-06-18:
You were probably using another emuMMC or normal sys.
And the migrator found an emuMMC, fixed the offsets and such and enabled it.

What did the migrator said that it found?
Maybe you had an old emmc backup on sd card and migration converted it to emuMMC.

Anyway check what actually you had and what happened.
(Also see what emuMMC are available in the change emuMMC section).

The whole issue here is basically confusion on what existed and what now exists and what is enabled.

raytan1807, 2020-06-19:
Hi, I would like to ask let say if I initialize my emmc (let say in switch A) and create a new emummc using hekate and transfer my emummc folder from my second switch (switch B), can I use the hekate migration option to make the games installed in switch B emummc folder to appear in switch A homescreen ? Thanks.

CTCaer, 2020-06-20:
You can do that as simple as what you are describing @raytan1807
They are encrypted and also expect specific customized info.

I8ikura, 2020-07-04:
The migrator says "Found SD Partiton based emuMMC! Do you want to repair the config for it?" I clicked continue. Then I went to "lanch" > "CFW (EMUMMC)" and I'm still getting "Unknown pkg1 version. I emuMMC corupt? Or Not yet supported HOS version! Failed to launch HOS! Press any key..." I'ma noob at this stuff so I don't know how to get it work.

CTCaer, 2020-07-04:
Well don't use v5.1.2..
There's v5.3.0 already
https://github.com/CTCaer/hekate/releases

CTCaer, 2020-07-19:
Closing as a non hekate issue

closed 2020-07-19 by CTCaer