CTCaer/hekate #460: Blue Screen after trying to launch EmuMMC

Dablestation, 2020-06-20:
I have followed the guides correctly and attempted it 2 times now. I have latest version of atmosphere and hekate. I create a partition for EmuMMC and do as the guides say. I can RCM and inject the hekate payload through Tegra and have access to the hekate menu. I created a EmuMMC in Raw1, but when trying to Launch->EmuMMC CFW, I get a blue screen crash. I have heard of someone else having this issue over at: #https://gbatemp.net/threads/switch-10-0-4-atmosphere-0-13-0-blue-screen.567840/

I can boot into custom firmware if I inject hekate through Tegra, but then go to payloads and select fusee-primary (instead of the launch option and EmuMMC).

CTCaer, 2020-06-20:
Did you update the bootloader folder or you only just inject the new v5.3.0 payload?
Atmosphere 0.13.0 did some changes and the old emuMMC driver does not work on that.
It creates a blue panic screen.

The fix is easy, download the latest hekate release and drag n drop the bootloader folder to your sd card root. If asked by your OS to merge and replace files, hit yes. (no need to delete old folder.)

Have in mind that when a new hekate release is done, you should always update the bootloader folder.

Dablestation, 2020-06-20:
I downloaded the latest hekate and dragged and dropped the bootloader as you said, but I keep getting the error. When downloading the latest hekate, what should be done with all the other folders? are they needed? Im speaking about bdk, modules, nyx, res, tools and all the other files inside the zip. Or just bootloader is needed? 

I am quite confused as I understand that as things get updated, guides become old, could this be an issue? Because following the guide I just downloaded everything from SDsetup and copied to the root of the SD card. Is this enough? or is there anymore to it that the guides fail to explain?

Also, have you encountered the problem I am speaking about before?

axionman, 2020-06-20:
Good morning,

I'm facing the exact same issue:
Brand new SD card, used the SDsetup website to get all files needed, but I get the blue screen everytime I try to launch the EmuMMC CFW.

CTCaer, 2020-06-20:
@Dablestation You downloaded the wrong zip.
The correct is this: https://github.com/CTCaer/hekate/releases/download/v5.3.0/hekate_ctcaer_5.3.0_Nyx_0.9.2.zip

What you should have pressed:
[image]

---------
@axionman That's strange. I will still investigate though.

Dablestation, 2020-06-20:
I have tried with that file too and it still doesn't work.

Answering @axionman comment, I haven't tried with atmos 0.12, but my emuMMC is a clean one and not a backup. I haven't touched the switch in a long time so maybe it has to do with firmware version.

@CTCaer Is there a minimum switch firmware version for the 0.5.3 release?

CTCaer, 2020-06-20:
Nope it should work with every version. And v5.3.0 should support 0.12 and older AND 0.13 atmosphere files.

Can you tell me what HOS version you have on that emummc?

Dablestation, 2020-06-20:
I am sorry for my ignorance, could you guide me on how to look at it? sorry...

CTCaer, 2020-06-20:
@Dablestation The easiest way is to use `Logs ON` from the Launch screen (top-right). It will show some text in a very fast fashion when you select what to boot, so you might need to get a video of it.

Then tell me the pkg1 version printed there.

Dablestation, 2020-06-20:
Here you go.

[image]

axionman, 2020-06-20:
> Answering @axionman comment, I haven't tried with atmos 0.12, but my emuMMC is a clean one and not a backup. I haven't touched the switch in a long time so maybe it has to do with firmware version.
> 

Actually my emuMMC was also a clean one, based my OFW system which is still on 9.01

CTCaer, 2020-06-20:
So I can confirm this on 9.0.1.
@Dablestation @axionman 
Please try this:
[hekate_test.zip](https://github.com/CTCaer/hekate/files/4808234/hekate_test.zip)

Dablestation, 2020-06-20:
I was literally going to post an update. I have redone everything and this time manually downloaded the files needed (following this guide -> https://nh-server.github.io/switch-guide/ ). I was advised on discord to not use SDsetup as it may be outdated, and from what I could see, SDsetup didnt merge the bootloader folders. I still get the blue screen. 

Will try the new file you have sent. Guide me at what exactly I need to do with the hekate.bin file inside the zip so we're on the same track please.

CTCaer, 2020-06-20:
You just inject it. and boot the emummc boot entry.

Dablestation, 2020-06-20:
Also, is it worth for me just to update the official firmware of the switch to a newer version and just starting there? Or do you advice against it?

Dablestation, 2020-06-20:
> 
> 
> You just inject it. and boot the emummc boot entry.

I dont have to replace anything inside the SD card?

Dablestation, 2020-06-20:
It worked! What is it that changed?

CTCaer, 2020-06-20:
Nope you don't need to replace anything. If you want to update to 10.0.X do it, but please after testing.

CTCaer, 2020-06-20:
Ah nice. It's a discrepancy with keyslots and new exosphere.

Dablestation, 2020-06-20:
So it is just a matter of replacing the hekate 0.5.3 bin with this one you sent me and thats it? Or there will be more tweaking to this test file before release?

Thank you so much btw. I will share on discord that the bug is 'resolved'. Do you want me to keep the file private until you post an official release? or is it stable enough to share with everyone that had this issue?

CTCaer, 2020-06-20:
Nope the file will be public until I push a new release.
Do not close the issue here in order to raise awareness for users with 9.0.x.
And I assume also 8.1.0.

You can use that as a daily driver until I make a new release, because it's based on v5.3.0 with only the fix added. If you want to share it, feel free. I will commit source change later.

Dablestation, 2020-06-20:
I will leave this open so more people can see it. I am glad it is on the right path for fixing. You do an incredible job. I will be donating shortly and others should do the same, the level of 'customer support' is unmatched. :)

CTCaer, 2020-06-20:
The fix: https://github.com/CTCaer/hekate/commit/9673d5524a2e6ce843cb4dd5b6ed668d6a526681
Binary: https://github.com/CTCaer/hekate/issues/460#issuecomment-647019768

Dablestation, 2020-06-20:
CTCaer, since you have been very helpful, I will ask you. If I launch hekate and go to payloads and load fusee-primary (instead of launch and FSS0 emuMMC), does it put me in emuMMC or sys? I read that theres the hekate.ini file you can edit in order to force emuMMC, but I am scared of breaking something.

CTCaer, 2020-06-20:
fusee-prim adheres to emummc.ini always. And any key like that in hekate ini, except icon and bootlogo, are only for hekate based launching of CFW (fss0= or kip1=) and not other payloads.
The moment hekate launches a payload, that's it. It controls nothing outside of itself.

i-am-the-slime, 2020-06-29:
I had the same problem and the "temporary fix" solved it.

CTCaer, 2020-07-19:
Fixed and released: https://github.com/CTCaer/hekate/commit/9673d5524a2e6ce843cb4dd5b6ed668d6a526681

closed 2020-07-19 by CTCaer