CTCaer/hekate #461: hekate v5.3.0 Crash to RCM after "Booting..." on 9.0.0 FW

SwiftMove, 2020-06-21:
hekate v5.3.0 is crashing to RCM mode with FW 9.0.0 installed. 

Happens regardless of Atmosphere version. I've tested on both Atmosphere v0.9.4 and v0.13.0.
I can boot to FW 9.0.0 using fusee-primary from Atmosphere v0.13.0 just fine. If I switch to FW 10.0.x, then hekate v5.3.0 boots just fine. But as soon as I downgrade to FW 9.0.0, it just crashes to RCM mode right after "Booting..." is shown on the display.

[image]

Ultracoolguy, 2020-06-21:
I can confirm this bug(I'm using 9.0.1). For me booting using fusee-primary worked, so if someone wants to boot into their CFW that can work in the mean time.

CTCaer, 2020-06-21:
This is a duplicate of the issue below it and also pinned.
It was fixed and a binary was supplied until the next official release. 
Check: https://github.com/CTCaer/hekate/issues/460#issuecomment-647019768

Blue screen means panic. It can react in 2 ways. Show the color or just reboot into rcm.

closed 2020-06-21 by CTCaer