CTCaer/hekate #463: SD File based emuMMC not working hekate 5.3.0 (AMS 0.13.0)

jshsakura, 2020-06-24:
[https://img1.daumcdn.net/thumb/R1280x0/?scode=mtistory2&fname=https%3A%2F%2Fk.kakaocdn.net%2Fdn%2FbsAbeN%2FbtqE7PkHuPH%2FlTGuJWMtyg5uTfQMzKqnL0%2Fimg.jpg](url)

First of all, thank you for making such a great program.
Currently in the latest versions of Atmosphere 0.13.0 and Hekate 5.3.0
A boot error occurs when using emuMMC based on SD File.

The payloed file used is fusee-primary.bin and the existing SD Partiton emuMMC runs fine without problems.
Is there a workaround for this?

CTCaer, 2020-06-24:
Well hekate has nothing to do with anything after it gets chainloaded to something else.
The chainloaded payload takes complete control of the boot process. In that case, fusee-prim.

In your case I would check if it works via booting with hekate first and then redirect the issue to the atmosphere repo.

CTCaer, 2020-06-24:
On the other hand I already checked v5.3.0 native boot compatibility with all firmware and the result is this:
v5.3.0 release:
- Boots all firmware with 0.12.0 atmosphere.
- Boots all firmware except 8.1.0 and 9.0.x with 0.13.0 atmosphere.

v5.3.0 fixed (https://github.com/CTCaer/hekate/issues/460#issuecomment-647030977):
- Boots all firmware with 0.12.0 atmosphere.
- Boots all firmware with 0.13.0 atmosphere. Including 8.1.0/9.0.x

So I will change the issue name but still wait for a validation from you.

KaimanDoro, 2020-06-26:
I am having the same issue as described; used latest Hekate, latest Atmosphere, file-based emuMMC, and fusee-primary is not launching.  Only difference is that I am using exFAT.  Both directly injecting and booting through Hekate fail.  Here is a screenshot:

https://i.imgur.com/bC5yiHd.jpg

Also this user on Reddit seems to be experiencing the same issue: https://www.reddit.com/r/SwitchPirates/comments/ek6ngh/cant_inject_fuseeprimary_black_screen/fvt6j63/

CTCaer, 2020-06-26:
Well this is the wrong repo to report that.
Any issues with fusee-primary should be reported to the atmosphere repo.

Basically, all of you need to understand that when payloads button or payload= in ini is used, it is like injecting that payload.

You can report issues here, **if** hekate's **native** boot method is used. And hekate's native method works.

jshsakura, 2020-06-27:
Thanks for the answer.

closed 2020-06-27 by jshsakura