CTCaer/hekate #467: SD-Card is initialized in 1-bit mode.

tegutistgut, 2020-07-03:
Hello,
I have a SanDisk 128GB Microsd card and I'm running the latest version of Hekate.
When I boot into Hekate this error shows up: https://ibb.co/3f0gHZ4
Here is another picture:
https://ibb.co/tQ210fZ

My sdcard is formatted in Fat32

Thanks in advance

CTCaer, 2020-07-03:
It's what you read on the pop up.
Your sd card reader is either not properly seated or a pin broke on the motherboard connector because of many ejections/insertions.

You can still use CFW and OFW. But the sd performance will be a mere 5MB/s until you repair that.

closed 2020-07-03 by CTCaer

tegutistgut, 2020-07-03:
So I can use cfw and emummc but it will be slow?

CTCaer, 2020-07-03:
Yeah. Extremely slow on emuMMC which uses SD exclusively.

tegutistgut, 2020-07-03:
Ok. Thank you very much for your help

tegutistgut, 2020-07-13:
So today I opened my switch and I saw, that a pin is missing (like on this video: youtu.be/F4aNuKR8Ies)
Is there anything I could do except what that guy in the video does?

impeeza, 2020-07-13:
Yes the only options are:

Repair the connector or use the alternative on vídeo.  a Hardware error can't be solved on software, if the SD doesn't have any of the Serial communications pins, will work on a 1-bit mode there is nothing to do.

tegutistgut, 2020-07-13:
Ok thanks

tegutistgut, 2020-07-25:
What if I use cfw sysnand? Would it be slow too if I play my dumped games (legally bought, of course) from my SD card?

Masamune3210, 2020-07-25:
Literally anything that access the SD card will be extremely slow. If it's on the nand it's fine for the most part as the system doesn't mess with the SD much other than cfw

tegutistgut, 2020-07-25:
Ok thanks

tegutistgut, 2020-08-01:
Would sx os work with the missing pin?
I remember that ctcear said that this problem could be fixed with a update of Hekate, gut he wouldn't do it.

impeeza, 2020-08-01:
It's a hardware issue no software can arround that.  The only way is get the hardware  fixed

tegutistgut, 2020-08-01:
Ok thanks

toastyishere21, 2021-06-24:
hello i use my ultra sandisk SD card fat32 im in 1 bit mode the sd card is speed is 6mb and it doesnt load cfw on the switch just blank screen when launched any possible ways to fix without buying another sd card if thats the case.

impeeza, 2021-06-24:
Read all the post. There is nothing to do brside repair the software. Hekate can not suppor anymore 1 bit mode

toastyishere21, 2021-06-24:
> Read all the post. There is nothing to do brside repair the so

 repair the hardware it surely isnt a software issue.

impeeza, 2021-06-24:
> > Read all the post. There is nothing to do brside repair the so
> 
> repair the hardware it surely isnt a software issue.

Damn keyboard and autocorrector, thanks @toastyishere21 for make me to see the typo

toastyishere21, 2021-06-24:
> > > Read all the post. There is nothing to do brside repair the so
> > 
> > 
> > repair the hardware it surely isnt a software issue.
> 
> Damn keyboard and autocorrector, thanks @toastyishere21 for make me to see the typo

i was looking through google and youtube mr. i saw that people factory reset their switches and said it would fix "faults" im amused that wouldnt work for the booting up cfw.