CTCaer/hekate #468: EmuMMC will not boot after update: "pkg2 decryption failed. Is sept updated?"

UnicornsOnLSD, 2020-07-11:
I updated my EmuMMC to 10.0.4 and I get the error in the title when I try to boot into it. My Atmosphere is 0.13.0 and my Hekate is 5.3.0

I've tried deleting /atmosphere, /sept, and /bootloader and starting with a clean Atmosphere and Hekate. That didn't work. I can still boot into SysNAND (10.0.0) with Atmosphere.

Here is my hekate_ipl.ini:

```
{------ Atmosphere ------}
[Atmosphere FSS0 EmuMMC]
fss0=atmosphere/fusee-secondary.bin
kip1=atmosphere/kips/*
emummcforce=1
icon=bootloader/res/emu_boot.bmp
{}
[Atmosphere FSS0 SYS]
fss0=atmosphere/fusee-secondary.bin
kip1=atmosphere/kips/*
emummc_force_disable=1
icon=bootloader/res/sys_cfw_boot.bmp
{}
{-------- Stock ---------}
[Stock SYS]
fss0=atmosphere/fusee-secondary.bin
stock=1
emummc_force_disable=1
icon=bootloader/res/stock_boot.bmp
```

One thing to note is that I accidentally mounted my EmuMMC partition in Linux, but that was after the issue started. I immediately unmounted it. Could this have messed up the partition by overwriting bits?

CTCaer, 2020-07-11:
Seems that something in the update didn't go as planned.
That error means that either pkg1 or pkg2 didn't copy properly or the emummc got corrupted (can happen if your sd card is exFAT and your emuMMC is file based and you didn't reboot when asked).

Can you enable the button in Launch to `Logs ON`?
It will show some boot logs on screen real fast. You will probably need to take a video of that. I will need you to tell me the pkg1 version (it's something that looks like a date).

UnicornsOnLSD, 2020-07-11:
Here's the output with logs enabled: https://youtu.be/PibPAL2_Pdc

```
Initializing...

Found pkg1 ('20200303104606`).

Identified pkg1 and Keyblob 10

Found FSS0, Atmosphere 0.13.0-32951329
Max HOS supported: 10.0.4
Unpacking and loading components.. Done!
Loaded config, pkg1 and keyblob
Generated keys
Loaded warmboot and secmon
Read pkg2
Pkg2 decryption failed!
Is Sept updated?

Failed to launch HOS!

Press any key...
```

CTCaer, 2020-07-11:
So your pkg1 is 10.0.x ("20200303104606").
That means that the issue is really on pkg2. It does not match with pkg1 or it's corrupted.

You will need to manually write the proper pkg2 or use a tool like https://github.com/suchmememanyskill/EmmcHaccGen to fix your emuMMC.

There are multiple guides on how to unbrick when an update goes wrong.
Next time you upgrade, it's also good to check the error messages before continuing (in case there's a missing file or something).

closed 2020-07-11 by CTCaer