CTCaer/hekate #469: Partition Manager Bug

talmagal, 2020-07-13:
When partitioning Android/Linux partitions at the same time it over rides linux and makes one big partition instead of two.

My set up :128Gb Sandisk 
HOS : 52Gb
emummc 29gb
Linux:16gb
Android 16gb

It shows properly at first but upon refresh it shows

HOS:52Gb
emummx:29Gb
linux 0GB
Android 32 Gb

When connect to pc via usb it shows 3 partition in disk management. 
The android mis sized/The Hos and the emummc partitions are both proper sizes

CTCaer, 2020-07-13:
That's not a bug. It's the magic of Hybrid MBR. If you connect it in a linux pc you will see what I mean.
As for the refresh thing, you read it wrong.
The information is not the same. The pop up shows raw stats about MBR partitions. Not if you have emummc/linux/android.
Just continue the process and flash linux, android, emummc and watch it magically work.

If you want to learn more of that process search for `hybrid mbr`, which is basically the combo of a working MBR and a working GPT. 
MBR is read by MBR only drivers (HOS, hekate, windows-when there's no only one partition table) and GPT by all 'nix based OS.

closed 2020-07-13 by CTCaer

impeeza, 2020-07-13:
Tagmal.  the GPT is a way to create partitions on a disk,  when you create a GPT table, also create a MBR one, that MBR is a only partition of all the disk, that is a measure for avoid old applications/S.O.s what do not understand GPT think the disk is empty and erases critical information.