CTCaer/hekate #470: (Request) Upload fusee secondarys contents in the hekate source code.

Aliasim800, 2020-07-18:
would please appreciate it if the contents of fusee secondary (kips, warmboot, exosphere and other contents) would be uploaded into the hekate source code.

Masamune3210, 2020-07-18:
....its an entirely separate project though, why would it be uploaded here?

Aliasim800, 2020-07-18:
Because fusee secondary is part of hekate and fss0 uses it which is also part of hekate and the main booting process hekate uses.

arch-box, 2020-07-19:
fusee secondary can be found here: https://github.com/Atmosphere-NX/Atmosphere/tree/master/fusee/fusee-secondary

Masamune3210, 2020-07-19:
Fuses secondary is not part of hekate, it's part of atmosphere, just like fusee primary....

bottmint, 2020-07-19:
You can easily get the contents of fusee-secondary by compiling atmosphere yourself.

Aliasim800, 2020-07-19:
thats the point, you should be able to get its contents without compiling atmosphere because its a big part of hekates boot process, maybe its not entirely part of hekate but fss0 uses it.

Masamune3210, 2020-07-19:
No. It's. Not.  

Hekate is a bootloader, it couldn't give to piles of manure what it's booting most of the time. Fss0 is part of the boot process because that's how the switch boots. It has nothing to do with hekate.  If you want easier access to a component of atmosphere, take it up with them.

CTCaer, 2020-07-19:
FSS0 is Fusee Secondary Storage. That's a component of Atmosphere and not hekate's.
It is used to pack every HOS boot time replacements. Security monitor, warmboot binary and kips (for now).

hekate is a bootloader that supports **natively** booting anything that is open source.
I have 0 reason to fork something only to increase the build times for hekate from mere seconds to an hour.
Especially when it takes you 10s to download and drag n drop both releases and done.
Or just change the makefile of atmosphere to pack all kips and stuff into a folder for easy copy paste.

closed 2020-07-19 by CTCaer