CTCaer/hekate #472: SD & eMMC data (@LBA 00540000) do not match! Verification failed.. Please try again.. Failed!

thien302466, 2020-08-04:
Hello! I followed https://nh-server.github.io/switch-guide/ to the letter, and when I tried to make a backup through "eMMC RAW GPP" it doesn't get past 9%! I am using a 256gb microSD card.

thien302466, 2020-08-04:
Also when I press the "launch" option, and select FSS0, it shows a message for a split second before rebooting back to hekate menu. The message says unknown pkg1 version.

impeeza, 2020-08-04:
Hi. Have you checked the sd using h2testw?

thien302466, 2020-08-04:
I am checking as of now! Will keep you posted!

thien302466, 2020-08-04:
Did you want me to test the WHOLE thing or just whats left?

thien302466, 2020-08-04:
Only 63033 of 98116 is being tested right now, appox. 1h 30m should be the time

impeeza, 2020-08-04:
yes if is big, can take lot of time, but you will be sure.  another question: is the SD formatted ExFAT or plain FAT?

thien302466, 2020-08-04:
The only formatting I've done was https://nh-server.github.io/switch-guide/user_guide/emummc/partitioning_sd/ labeled "Format for emuMMC setup . I had to choose this because I had no idea how to update my cfw and other things in 2019, so all it did was delete everything...? Since I didn't know how to do anything, I just started copying and pasting randomly trying to replace things, so the last option I had was to delete everything... Then I followed this guide to reinstall everything, and here I am, with just these two problems: the mismatch, and the unknown pkg1 version.

CTCaer, 2020-08-04:
Your sd card is probably fake.
Send me a screenshot (3 finger tap) or a photo Console info -> Sd card.

thien302466, 2020-08-04:
How does one do that?

CTCaer, 2020-08-04:
>Send me a screenshot (3 finger tap) or a photo **Console info -> Sd card**.

thien302466, 2020-08-04:
I still don't know where that is... Unless you mean't data management?

CTCaer, 2020-08-04:
In hekate menu.
[image]

thien302466, 2020-08-04:
OH! Of course.

thien302466, 2020-08-04:
[image]

thien302466, 2020-08-04:
Ah, so it was 128gb

CTCaer, 2020-08-04:
Yeah it's fake
In red, the info that gives it away.
[image]

closed 2020-08-04 by CTCaer

thien302466, 2020-08-04:
So I need an actual one? This links to both problems?