CTCaer/hekate #473: Error using USB tools

Matt3oV, 2020-08-06:
Hi, I'm trying to mount my sd card with UMS but as soon as I select it (or click any other option of the USB tools menu) I get after a few seconds an error that says "Timed out or canceled", my pc is running windows 10 2004 and my sd card is a samsung evo formatted in exFAT, I tried to add the reg file for windows but noting changed, also tried different USB ports and cables.

CTCaer, 2020-08-06:
That can happen for 3 reasons:

- Bad USB port (switch or pc)
- Bad cable
- Your PC usb hub rejects it for power reasons. Try to remove a device from the same group of ports.

Matt3oV, 2020-08-06:
Ok, I have done some testing, disconnected all the usb devices from the pc and tried all the ports, tried 3 different cables and tried on a different pc altogether, still same issue, the only thing that's left to test is the USB port on the switch, although I can use memloader without any issues so I doubt the problem is there, any ideas on how to check if the port is ok?

CTCaer, 2020-08-07:
What type of switch you have? And how you inject hekate?

Matt3oV, 2020-08-07:
I have a V1 unpatched switch and I inject hekate via the sx os boot menu

CTCaer, 2020-08-07:
Well there's the issue.
Anything that is sx has broken hw inits, they mangle the memory carveouts and more.

Replace your boot.dat with the one inside this zip:
[boot.zip](https://github.com/CTCaer/hekate/files/5042519/boot.zip)

Closing as invalid issue (bad chainloader mangles environment)

closed 2020-08-07 by CTCaer

MartiniMoe, 2021-05-03:
Hello, I'm using a v1 patched switch with SX Core modchip and I just stumbled upon this issue, because I have the same problems with UMS.
Can I also use the boot.dat from the linked zip? May I ask what exactly is the boot.dat in there?

Thank you very much :)