CTCaer/hekate #474: Battery drains?!?!

azoreseuropa, 2020-08-10:
I never had that issues months ago until I decide to download the latest atmosphere version and Hekate version. When I didn't used it then I turn the power off through Hekate because I am in autoRCM. When I turn it on through ns-usbloader and noticed that my battery drains already and it is zero. I have to recharge.. Twice! It was never happened to me when I had old atmosphere version and old Hekate version. Now the newer versions caused my battery issues. I don't know if the battery issues is relate to atmosphere or hekate ? HELP ?

CTCaer, 2020-08-10:
As I explained to you in discord:
>the issue is that you connected a usb or you pressed power by mistake
this wakes the console and enters in autorcm which drains the battery
so when you select power off, you should be careful to not press PWR or connect/disconnect a powered cable
basically you just described the classic "device woke up in autorcm and drained the battery" thing

closed 2020-08-10 by CTCaer

azoreseuropa, 2020-08-10:
> As I explained to you in discord:

Thank you for your replying but I am not sure I understand you. What do you suggest I should do ?

impeeza, 2020-08-10:
Hi. You should go to discord and continue your questions there, this space is for REPORT TRUE ISSUES no for ask for help

azoreseuropa, 2020-08-10:
> Hi. You should go to discord and continue your questions there, this space is for REPORT TRUE ISSUES no for ask for help

Hi. I am not sure which discord list should I talk to about it ? I forgot. I have many lists in discord, lol.