CTCaer/hekate #475: Making Pin, would this solve the problem?

tegutistgut, 2020-08-26:
Hello. 
I'm sorry for disturbing, but I want to be sure that I don't make things worse.
I have the same pin broken like on this video (www.youtu.be/F4aNuKR8Ies) and I want to ask 1 thing:

If I would follow the steps on this video (www.youtu.be/F4aNuKR8Ies), would my problem get solved?

TIA

impeeza, 2020-08-26:
This was already asked. And closed because is not an issue of software. Is a hardware issue and should be resolved on hardware. Please DO NOT OPEN ISSUES FOR ASKING SUPPORT. USE THE CORRECT SUPPORT CHANNELS

closed 2020-08-27 by tegutistgut

CTCaer, 2020-08-28:
>DO NOT OPEN ISSUES FOR ASKING SUPPORT. USE THE CORRECT SUPPORT CHANNELS

@impeeza do not act like that here. Even if you are right about what it is.

--------------------
@tegutistgut Yes this will fix your setup.
And yeah here is not a general support forum.

impeeza, 2020-08-28:
Hi, You are right, I don't mean to be so harsh, I Apologize.  Regards