CTCaer/hekate #478: Missing GPT support

EthanWeegee, 2020-09-06:
I copied the partitions from an SD card I use with hekate to a larger one, having recreated the partition table as GPT. To my surprise, hekate doesn't believe a fat32 partition exists on the SD card. Reading other issues, this seems to be because GPT simply isn't supported. Do I understand this correctly?

Conan179, 2020-09-06:
Why GPT? MBR working to 2tb.

CTCaer, 2020-09-06:
GPT is also not supported in HOS for SD.
Convert it to MBR and use hekate to partition it.

Have in mind that I will probably never add actual GPT support, even with a warning that this card will not boot CFW.
Reason is that the driver update will make the main binary bigger.

closed 2020-09-06 by CTCaer

EthanWeegee, 2020-09-06:
Why GPT? It makes more sense to use, and is more common.

I didn't plan to use Atmosphere, I'm running Linux.

CTCaer, 2020-09-06:
Ask nintendo.

In linux it makes sense but it doesn't matter. Also the bootloader still needs to impose such restriction to minimize issue tickets on why it boots hekate but not CFW.
As for android, for that reason hybrid MBR exists.