CTCaer/hekate #479: Switch black screen after injecting payload

NotLinky, 2020-09-12:
My switch died while in rcm mode so i charged it for about 2 hours, but when i try to inject the payload the switch screen stays black, with and without sd card, and i tried it with multiple cables and computers, the switch is not patched i know that for a fact. Can someone please tell me a solution to my problem? Also when i inject the payload tegrarcmgui  just stays on a green checkmark

impeeza, 2020-09-12:
You must:

> No open a issue ticket looking for support. For that are others channels.

> before try to inject the payload press and keep pressed the power button for at least 20 seconds then try the injection process again

NotLinky, 2020-09-12:
> You must:
> 
> > No open a issue ticket looking for support. For that are others channels.
> 
> > before try to inject the payload press and keep pressed the power button for at least 20 seconds then try the injection process again

I have pressed it for 20 seconds but still didnt work

Masamune3210, 2020-09-13:
If the switch dies when in rcm, it takes forever to charge up to where it's functional. Just leave it on its charger for around 3-4 hours and then try to push the payload again. Either way this has nothing to do with hekate

CTCaer, 2020-09-13:
Do you see black screen with battery icon, or just nothing?
Also are you using the latest version (5.3.2)?

impeeza, 2020-09-13:
I was about to ask the same, did the charging icon pops?  if not, leave charging for several hours.

NotLinky, 2020-09-13:
> I was about to ask the same, did the charging icon pops? if not, leave charging for several hours.

No it did not pop up and i have been charging it since i posted it and i tried it again but nothing still

impeeza, 2020-09-13:
are you sure your charger is working well?  use another USB-C charger or cable.  if you have a USB wattage meter will be useful.

NotLinky, 2020-09-13:
> are you sure your charger is working well? use another USB-C charger or cable. if you have a USB wattage meter will be useful.
Yeah i tried it with 2 switch chargers and then the  pro controller charger and some usb c to usb a chargers and it is fully charged but it still wont turn on

CTCaer, 2020-09-18:
Try this https://cdn.discordapp.com/attachments/595221592003575844/710925018514587708/rcm_check.bin

NotLinky, 2020-09-18:
> Try this https://cdn.discordapp.com/attachments/595221592003575844/710925018514587708/rcm_check.bin

did nothing at all

CTCaer, 2020-09-19:
Then your battery seems dead. Try to re-plug it.

There's nothing more I can help with basically. That rcm_check payload is ultra minimal and only allows you to see that everything is ok. Doesn't use any sd card or any other "heavy" hw other than display.

closed 2020-09-19 by CTCaer