CTCaer/hekate #480: failed to launch hos because exfat

zsw12abc, 2020-09-15:
i faced this issue after i update system to 10.2.0

sd card is exfat and installed hos driver only supports fat32 failed to launch hos

i check that i may need a fs driver to support exfat.
where can i get it?

FezVrasta, 2020-09-15:
Having the same issue, I tried to update Hekate and Atmosphere to latest but I'm still getting this issue. The OWF boots just fine though.

CTCaer, 2020-09-15:
This is a message that normally checks if your HOS update is the FAT32 only or the FAT32 + exFAT one.

In this case though, it's a bug because FS kip hashes are not yet in and hekate thinks you have the fat32-only update.
Will be rectified later this day.

aceadamuk, 2020-09-15:
I have same issue, will i just need to download the release to get this fix? thanks

CTCaer, 2020-09-15:
You will see a release in a couple of hours.

CTCaer, 2020-09-15:
Fixed in https://github.com/CTCaer/hekate/commit/8f7bdd45b15cca7c20ffaf6e27e0ab2665ad040d

Please download the new release: https://github.com/CTCaer/hekate/releases/latest

closed 2020-09-15 by CTCaer

kenson96, 2020-10-03:
hi I have same issue, can guide how to fix? thanks

aceadamuk, 2020-10-03:
> hi I have same issue, can guide how to fix? thanks

Update everything, Atmosphere and bootloader