CTCaer/hekate #485: Weird Big unreadable Font

popy2k14, 2020-09-19:
Hi, using newest "hekate - CTCaer mod v5.3.3 & Nyx v0.9.4" with an 512 GB Sandisk MicroSD Card.
Mostly on boot i now get the attached unreadable font with an message in the foreground.
I am booting the payload directly with my internal trinket.
Updated a few days ago from 5.3.2 and had hopes the issue is fixed.
Sadly it's not.

Any hints what the issue can be?
Windows does'nt find any errors on the micro SD card.

PS.: Upgraded 2 months ago from an 400 GB Sandisk MicroSD to an 512 GB.

thx and keep up the good work.

[image]

popy2k14, 2020-09-19:
Also got the following errors: Warning: Nyx is missing! & Failed to init sd card

Also tried/checked the following:
- hekate payload always gets found from my trinket chainloader and gets booted (most of the time without nyx)
- replacing the payload.bin (chainloaded from trinket) with any other payload always loads the payload correctly
- replacing the payload.bin (chainloaded from trinket) with SXOS one works correctly and brings me straight into SXOS or Stock Firmware when i am interrupting SXOS bootloader with vol+ key and press stock firmware

When booted to stock or SX anything working as expected. Official downloaded/bougth games like Suoer Mario Odyssey loads normally and plays normally.

How can i get logs when i cant input something on the touch because i cant read anything, neither sometimes nyx isnt loading?
How can i help you debugging?

r4dius, 2020-09-19:
Got the same thing once after upgrading hekate / nyx 5.3.3, powered the console off for half hour before looking for the problem but then it booted correctly ..

popy2k14, 2020-09-20:
yeah, when i shut it down with long power press and restart it, sometimes it works.
Now gone back to the SXOS payload for an workaround, that always works.

How can i help to debug this issue?
Seems an microsd reading/timing issue because trinket chainloader, SXOS & HOS can read the SD without an issue!

PS.: FAT32 formated

r4dius, 2020-09-20:
Mine is xfat, btw this morning it won't run for the first time again, got a nys is missing message, then the nyx with scrambled like your screenshot again, I put the sd on pc and transfered sept and hekate files again, looks like it's always ok,
I used the ftp from sxos the first time I updated and the problem occured...

popy2k14, 2020-09-20:
I have not updated hekate with ftp or something like this.
Updated with pc and also tried to recopy the bootloader directory without success.

r4dius, 2020-09-20:
Did you update sept too ?

popy2k14, 2020-09-20:
Yes, sorry forgot to say this.
Hrkate + nyx + atmosphere incl Sept.

r4dius, 2020-09-22:
Ok still randomly buggy for me too.., will try redoing the sd to check

popy2k14, 2020-09-23:
Any hints?
Is the same sd card working after redoing/reformating everything?

CTCaer, 2020-09-26:
Can you drag n drop the bootloader folder from v5.3.**2** or v5.3.**1** release and tell me results?
No need to inject an older payload. Just drag n drop the folder. It will throw an error about dragons, it's ok.

Basically the test here is to check if a change in Nyx v0.9.4 created that issue.

popy2k14, 2020-09-26:
Test the following with clean bootloader dirs (stright from the zip's):

- hekate 5.3.3 + Nyx v0.9.4 -> Nyx is missing
- hekate 5.3.3 + Nyx v0.9.3 -> Nyx is missing
- hekate 5.3.3 + Nyx v0.9.2 -> Nyx is missing

Switched to sx os payload after i created this ticket, this always works.
Now it seems something changed, because now i dont get this weird big font sometimes, i ALWAYS get -> Nyx is missing.

The microsd is good (tested on 2x windows PCs , Android & SX OS on the switch).

Is there anything i can do to help debugging?

thx

popy2k14, 2020-09-26:
Tried another/older fresh FAT32 formated MicroSD Card JUST with hekate -> booted 10x times in hekate with Nyx without an issue.
Will now backup and redo/reformat the 512 GB Sandisk MicroSD card. Will come back here.

CTCaer, 2020-09-26:
Take a photo of Console info -> SD card. Either via Nyx if it loads or from text UI.

popy2k14, 2020-09-26:
Have done the following:

- "diskpart clean" on windows
- recreated 1 exFat partition with windows storage manager (just to have the partition)
- reformated the sd with guiformat.exe (fat32formatter with gui) with fat32 and cluster site of 32kB
- recopied just hekate + atmosphere + Nintendo folder
- fixes archive bits like described here: https://gbatemp.net/threads/how-to-fix-archive-bit-for-all-sd-files-and-folders.515258/
- boot 10x hekate 5.3.3 + Nyx 0.9.4 without an issue
- booted into hekate -> stock sysnand -> started mario odyssey -> Works

Here is the SD info:
[image]

Can you see something special about my card?

Will now copy back custom fw and see how it goes.

PS.: in the beginning (when i got this SD all worked normally (on fat32) like now), the issues just popped up a few days after updating hekate. Dont want to blame hekate :-) just reporting how the issue popped up.

CTCaer, 2020-09-26:
Your card is legit. Other than that it's one of those sandisk U1 cards, it should not have an issue.
Also I don't mind blaming hekate. I prefer catching an issue if it really is an issue.
But if the issue was really hekate, I would have the github issues here filled with that.

The ultimate test is to get a emmc backup. Then exactly after, check this section:
[image]
If not all are zeroes or fails, reopen the issue.

closed 2020-09-26 by CTCaer

popy2k14, 2020-09-26:
Copied back my stuff and booted stock 10.1.0 without issues multiple times from hekate.
Done the backup, and no errors:
[image]

Then booted one time sx os 3.0.5, powered it off and want to load hekate:
[image]

Switched to sx os payload again -> this time got black screen also on sx os sometimes on boot. Seems completly random :-(
Updated sysnand and emunand to 10.2.0.... nothing changed.

Please see this thread here: https://gbatemp.net/threads/sx-os-3-0-5-boot-hangs.573854/
This post also had the pixalated issue: https://gbatemp.net/threads/sx-os-3-0-5-boot-hangs.573854/#post-9202664

Maybe some sx os <> hekate relaited stuff going on?

CTCaer, 2020-09-26:
You should have told me that you chainload like that.
I thought you just used their dongle.

Anyway that piracy thing has a completely broken hw init.
Also further discussion about sx is not allowed and also not my issue.

popy2k14, 2020-09-27:
Hekate dont work when i'll: Power on my switch -> trinket loads HEKATE payload directly from sdcard.
So no chainloading from piracy thing bootloader :-)

impeeza, 2020-09-27:
Hi, but you issue appears after use SX-OS so there is no way you receive support of that on the open code community.  Any support must come from they. You pay for the software THEY MUST GIVE THE SUPPORT

fgmelo, 2020-10-23:
Same happens with me

domtaquito, 2021-11-08:
Let me try and save someone time if they come across this again...
The EXACT same thing just happened to me and I solved it no less than 5 minutes from writing this comment.
I upgraded to a new 500GB SD card and had the issue. I compared the files from my old SD card and realized there was only 1 difference apart from other switch folders...
I had the payload.bin file in the root of the SD card. It needed to be in /bootloader/payloads/
Moved the file there and was able to push Hekate no problem. No more weird text glitch thing.

popy2k14, 2021-11-08:
In my case the glitch never happend after ditching SX-OS alltogether and just use ATMO.
So the issue clearly popped up after booted SX-OS one time and after that hekate.
So for me issue is solved.

SensitiveBug, 2022-05-01:
I also ran into this issue.  It happened after installing eMMC but I think I also ran into it before once. (I only own a switch for a few days)
It turned out my memory card is a fake one and got corrupted every time.  I had a 256GB card which turned out to be only 32GB.

captain-black, 2022-08-06:
I meet the sam problem, Sandisk 512G U1 A1 SD, It can boot into Hekate, but shown unreadable font. 
I'm using hekate 5.8.0 + nyx 1.3.0

captain-black, 2022-08-06:
oh! I fount it! It's a microSD card slot problem...the connector is loose contact! But my 64G SD card is still working so well.
So weird!
I opened my switch shell, make a filler-piece to press the connector, it works with my 512G SD now!

CTCaer, 2022-08-07:
If nyx loads and fails to load resources from sd (like fonts), it almost always is a sd reader connection issue.
That one sd works and another doesn't is because you move around the connector when you eject/insert it. So it's pure luck there.

captain-black, 2022-08-08:
My 64SD always works, but 512 SD not!
like this video: 
https://www.youtube.com/watch?v=F4aNuKR8Ies