CTCaer/hekate #486: Emunand CFW and Sysnand stopped working after sysnand restore

MatteKalle, 2020-09-19:
Hi,
After putting myself in a situation where nothing is working anymore I am desperate for help. I have run sysnand CFW for several years. The goal now was instead to achieve emunand on SDcard and OFW on sysnand. Also hopefully without having to reinstall games or loosing any saves. I want sysnand OFW for being able to play online.

Setup before begining:
Unpatched switch.
NTFS formated sd-card
Firmware 10.1.0
Atmosphere 0.13.0
The error message about game cart slot appearing now and then.
Auto RCM
Probably some burnt fuses

So this is how it went down. I hope I remember to write all the crutial steps and the order which i did them in.

1. I located the necessary rawnand.bin, boot0 and boot1 for making the nand restore. This is a backup I made before going CFW. I dont know what firmware version I was on when I made this backup but I can see that the files are dated december 2017.

2. I downloaded hekate_ctcaer_5.3.2_Nyx_0.9.3 and put the bootloader folder on the SD card root.

3. I transferred a copy off all my SD files to my computer.

4. I started hekate by using hekate_ctcaer_5.3.2.bin.

5. I used "partition SD Card" in Hekate and created a HOS (FAT32) and emuMMC (RAW).

6. I created emuMMC SD Partion in Hekate and it says Enabled!.

7. Copied back all the backuped SD card files to the root off the SD card.

8. Tried to launch by entering RCM-mode and injecting fusee-primary.bin. It worked and I checked system firmware and saw that I was on E (emunand).

9. Tried going in to hekate again to launch from there but at first it did not work. Tried a bit and I think what got it working was when I added fusee-primary.bin to the payloads folder in bootloader folder (it was empty before) and also adding code to the hekate.ipl.ini. The code I added was some lines found in a forum thread. One for starting CFW (EMUNAND) and one for Stock (SYSNAND).

10. Renamed the hekate_ctcaer_5.3.2.bit to reboot_payload.bin and overwrote the file in the atmosphere folder (for being able to reboot to hekate).

11. Copied the backuped Nintendo-folder to emuMMC\RAW1 and comfirmed that my installed games was working again on emunand.

12. Now when emunand was working I was convinced it was safe to restore my nand backup. I followed this guide https://suchmememanyskill.github.io/guides/nandrestore/ and did not get into any trouble.

13. Entered RCM-mode and injected (the same) fusee-primary.bin as before sysnand restore but now nothing happens. The screen is black and TegraRcmGui shows that it is ready to inject again after like 2 seconds.

14. Trying to inject hekate_ctcaer_5.3.2.bin again and it works.

15. Trying to launch CFW (emunand) through Hekate exactly as I did before restoring the sysnand. Now it seems to start up, it shows the Atmoshpere logo, it shows the Nintendo Switch logo, it shows the battery status symbol and then for a short moment the sleep screen (the screen where you have to press a button 3 times for it to wake up) but then the Switch goes to black screen and I see that Tegra rcmGUI shows that it is ready to be injected again.

Usually the error about the game cart slot comes up when the switch starts up so I was thinking that it might have something to do with nogc. But I tried fiddling a bit with the .ini-code turning it off and on but it does not seem to make any difference. The console keeps crashing.

When trying to boot sysnand trough hekate I get Unknown pkg1 version. Hos version not supported! Failed to launch HOS". But I have not yet looked in to this any further. Step 1 seems to be to get emunand CFW working again.

So what happend? I can't understand how I could go from from working emunand to not working emunand just by restoring sysnand. Any thoughts and recommendations for further troubleshooting are highly appreciated.

Please tell me if you need to know the ini-file code or anything else.

masagrator, 2020-09-19:
> NTFS formated sd-card

You sure? It should be exfat or fat32.

MatteKalle, 2020-09-19:
Sorry, my bad. Off course it was exfat before.

impeeza, 2020-09-19:
That could be a issue. If you install a firmware in a exFat SD and convert it to FAT32 (Hekate uses FAT32). The firmware do not work well

MatteKalle, 2020-09-20:
Thank you for trying to help!
I have not installed any firmware first and then converted. It was the other way around. First partitioning with hekate = fat32, then emummc.
Can you please elaborate a bit? I can’t understand how that could have affected anything.
To begin with I suppose you are talking about the emunand not working, right?
If your theory are right how can it be that emunand was working before I did the Sysnand restore. That should not have affected the emunand, right?
Also it seems odd that I have seen no warning anywhere about going from exfat sd to fat32 when creating emunand.

impeeza, 2020-09-21:
Hi,  Sorry for my English, it's is no my native language.

In the past, someone had a trouble because he update his system to a new firmware, and then replace the SD but change the format of partitions between exFat and FAT32 because when you select the firmware to install you must to select if the SD is on FAT32 or exFat.  he regain control over his system migrating the data to correct format.

MatteKalle, 2020-09-21:
No problem, It's the same for me =).

Are you reffering to the problem that I am having with sysnand, emunand or both?
Either way I can't understand how the above would apply to my situation.

Before I started tinkering with this I was on sysnand CFW with exfat SD. Then I created emunand from the sysnand with Hekate. In that process it changed my SD card to FAT32. But after this I could start up the emunand and play my installed games exactly as before. It was not until after doing the sysnand restore that the trouble with both sysnand and emunand begun.

Anyway, If you still think that the SD format may be a problem. What do you suggest that I do? Should I backup the stuff on main partition on SD card and reformat it to EXFAT and then transfer the files back? I suppose I should not touch the hidden emunand partition?

MatteKalle, 2020-09-24:
I've tried formating the main partition to exfat but that did not do any difference so I formated it back to fat32.

But after testing some more I finally reached some kind off breakthrough.
I started up emunand as stated before but when seeing the sleep screen (after the battery symbol) I started spamming the home button. When I did that I got the game cart slot error message and after pressing OK on that I could access my Switch and start up games. Although if I press the power button once the Switch dies again.
So there seems to be some kind off crash when it goes inte sleep mode, right?
Now that I at least now a way to get in I'm thinking off updating to the latest (hackable) firmware to see if that solves the problem.

I'm still welcoming all suggestions and thoughts about this unique (?) situation that I've gotten myself into =).

MatteKalle, 2020-09-25:
Updated emunand to 10.2.0 but the same problem remains.
Would really appreciate some help figuring this out.
If nobody is able to point me in the right direction I suppose my next step will be to try to repair the boot0 and boot1 and also test to backup and restore everything to another sd card.

CTCaer, 2020-09-26:
Well you changed your boot process to fusee-prim so I can't help with that. Also when you reach inside Atmosphere you get fatal reports. You should request help in the atmosphere repo for the emummc issues.

As for sysnand where you use hekate to boot, that message shows up when it can't identify the BOOT0's bootloader, named package1.

You either restored something that was problematic or you inject an old version of hekate. Make sure that you see v5.3.2 or v5.3.3 in Nyx (hekate's GUI).

MatteKalle, 2020-09-27:
Thank you for taking the time to answer and sorry if I'm posting in the wrong place.
Because I can't find any similar posts any where I'm reaching out where ever I can.
I will try writing in the atmosphere rep. about the emunand problem then.

Do you have any suggestion regarding the sysnand? I am using 5.3.3 hekate and I can see that in Nyx, Then it must be the BOOT0 right? Can I repair it in some way?

CTCaer, 2020-09-28:
You can repair everything as long as you have a copy of your prodinfo (cal0).
There are many great tools and guides out there where you can choose a firmware and rebuild your emmc and unbrick it.

MatteKalle, 2020-09-28:
I have a file named prodinfo.sysnand.bin on the root off my SDcard and in the backup folder on my computer I got a file named prod.keys.
Is it any off those that you are referring to?

MatteKalle, 2020-09-30:
So before searching for more info about the prodinfo file and the repair I tried something more simple.
I tried backing up my emummc with Hekate (boot-files + rawnand).
Then I used another SD card and this time made the partitions with tegraexplorer.bin.
After that I placed the latest atmo and hekate and started with the latest hekate bin-file.
In Hekate I Created the emuMMC as SD Raw Partition.
I then restored the boot-files + rawnand to the SD-card.
When I now try to boot up cfw emummc I get hekate boot logo and then it goes directly in to RCM again.
After that I copied everything (except emummc folder) from my other SD card.
Tried again but the same thing happens when trying to go to emuMMC.
When I put in my other SD card again it is working as before = emuMMC boots but crash at sleep.

Any thoughts about this and/or the prodinfo question is welcome.

CTCaer, 2020-09-30:
I don't recognize the naming of prodinfo.sysnand.bin but you can check if it matches PRODINFO file if you try to do a SYS backup of your sysnand.

Other than that, reboot to rcm means that a fatal happened. You might want to check that emmc and sd card info in Nyx report proper values.

MatteKalle, 2020-10-01:
I am sorry but I don't understand what you mean when you say "do a SYS backup of your sysnand".
I thought thats what I've done when I backuped the boot0, boot1 and rawnand.bin.
What kind off backup is it that I have to do to get the prodinfo file?

MatteKalle, 2020-10-03:
So can you help mee check the values if I post a picture off my "Internal eMMC info" and also "microSD Card Info". Should I blank something out? Like serial number.

MatteKalle, 2020-10-05:
Finally got sysnand working by getting live support from a kind person on Discord.
My backup off boot0 was faulty so that was the problem. When we looked at the file in a hexeditor it only showed zeros.
He helped me updating my firmware and to repair and restore my boot0.
Sysnand with online working as it should now =).

Next step will be to get emunand working. Will see if I can achieve that without needing to reinstall my games.
I'm thinking that I will start by doing a backup off my working boot0, boot1 and rawnand.bin.
Then I will try to restore only boot0 and boot1 to my emunand to see if that fixes the sleep error in emunand.
If not I suppose I will have to backup my game saves and create a completely new emunand from my working sysnand.
Please advice if you have any advice regarding this.

MatteKalle, 2020-10-06:
So to close this thread up I can tell you that emunand is working again.
I did not have to do anything more to get it working. Obviously the sysnand fix also fixed the emunand sleep problem.

CTCaer, 2020-10-07:
Strange that emummc was even able to run with a faulty boot0.
Note down the steps of unbricking you learnt. These will help on any fw update/restore gone bad.

closed 2020-10-07 by CTCaer

MatteKalle, 2020-10-07:
I don't think that it was a faulty boot0 on emummc. The faulty boot0 was from the backup I did before I hacked my Switch. When I restored that to sysnand It stopped working. It also meant that emummcs sleep mode stopped working.
When fixing the boot0 and updating firmware on sysnand, emmummc sleep mode also started working again.
So I suppose the sleep mode error had something to do with sysnand being on a low firmware and burnt fusees. Just guessing from some things that I read about when looking for a fix.