CTCaer/hekate #487: Unknown pkg1 version. Failed to launch HOS!

squimples, 2020-09-21:
I've never done this before and I do not know what I'm doing. I believe I've got hekate on my switch but when I try to launch, then press CFW (I'm following along to a youtube video) the error message "Unknown pkg1 version. Failed to launch HOS!" pops up. I've tried reading other posts but I don't understand what to do or what I did wrong

impeeza, 2020-09-21:
RTM 8D

squimples, 2020-09-21:
can i just restart the process or is there something I should do first?

CTCaer, 2020-09-26:
Make sure you have the latest hekate.
Assuming that this is your first time running CFW and not in emuMMC, this error would be thrown if you have the latest Switch OS version and old hekate.

closed 2020-09-26 by CTCaer