CTCaer/hekate #489: sd & emmc data (@lba 00088000) do not match

ItsRoryG, 2020-09-23:
Having the following issue when restoring eMMC recently - getting error code sd & emmc data (@lba 00088000) do not match. I have been using this SD Card without issue for around 3 months. I get to around 13 minutes and I get the sd & emmc data (@lba 00088000) do not match verification failed issue, running on the latest version of Hetake.

SD Card is a 64GB Kingston card. I suspect its an issue with the SD card itself as I use SanDisk for my other 3 switches. 

Anything I can do here?

CTCaer, 2020-09-26:
Your card is probably fake.

Take screenshot from `Console info -> Sd card` (either tap with 3 fingers or press the capture button on jc or take a photo) and up it here.

ItsRoryG, 2020-09-26:
@CTCaer SS as requested:

https://i.imgur.com/Z2al66q.png

Would the issue be caused by it being exFAT? As I mention it worked fine for 3 months then suddenly the issue appeared.

CTCaer, 2020-09-26:
Card seems legit. And no, exfat should not affect the behavior. At least inside hekate.
That issue can only appear if sd card is fake or sd card reader has issues or there's some filesystem corruption.
Try to check the following section after running backup again and failing.
[image]
These are indicators of issues of the first 2 reasons.

Other than that, there's nothing more I can do to help pinpoint the cause.

closed 2020-09-26 by CTCaer