CTCaer/hekate #490: Hekate ipl

mbamber1986, 2020-09-28:
Hi i have 2 emummc setup on my switch with different firmwares is there anyway without manually changing in emummc option that j can make hekate ipl forward to say /emummc/sd01 also another entry for raw01 while leaving raw default if I boot with fusee primary 

Sorry for a random request

CTCaer, 2020-09-28:
Yep, use hekate to boot.
Allows you to change emummc on the spot by tapping a different boot entry in Launch.
`emupath=` is the trick. You can read readme to see what needs to be written there and how it works.

closed 2020-09-28 by CTCaer

mbamber1986, 2020-09-28:
Thanks i also sorry I also use fast cfw switcher is it possible to add the option it has options of boot positions