CTCaer/hekate #492: Switch unresponsive after prompting for entry of date and time for Nyx

BalletDuckNinja, 2020-09-28:
Joycons and touchscreen work on hekate 5.2. Updated to 5.3.3 and upon payload injection I'm prompted to enter date and time. Both the touchscreen and joycons don't work. Tried downgrading to 5.2 and both work on nyx again.

BalletDuckNinja, 2020-09-30:
On fresh install of 5.3.3 files, there's a brief period where the switch responds after injecting hekate before it becomes unresponsive. I could quickly confirm the date and time for nyx, but it immediately froze afterwards. Subsequent injections immediately cause an unresponsive switch as well.

CTCaer, 2020-09-30:
That's a different thing than what was discussed in discord?

BalletDuckNinja, 2020-10-01:
I'm not sure if it's related because I haven't seen any SD card error messages in days of injecting and removing/reinserting the SD card, but injecting hekate 5.3 and up always causes a freeze. Emunand also runs at normal speed if that's any indication.

CTCaer, 2020-10-01:
The 1bit mode message is basically like the pre-final straw getting broken.
The sd controller can report a working sd and then at the next reads it can fail or semi-fail and continue.
Reason why the error counts in sd card info and errors in the console window exist.

If you can reliably cause the hang, then that makes the situation even easier to debug.
Ping me in discord when you have time or to tell me when you will have time.

CTCaer, 2020-10-28:
After communication with user, the issue stems from: [nyx: Reduce idle power draw by 280mW](https://github.com/CTCaer/hekate/commit/5790c585f59473dc3019562450bbaafd0f177ed6)

Constant changing of DRAM frequency causes a hang. (Which normally points to a hw issue.)
Investigation is needed.

chispas90, 2020-11-20:
> After communication with user, the issue stems from: [nyx: Reduce idle power draw by 280mW](https://github.com/CTCaer/hekate/commit/5790c585f59473dc3019562450bbaafd0f177ed6)
> 
> Constant changing of DRAM frequency causes a hang. (Which normally points to a hw issue.)
> Investigation is needed.

I'm having the same issue than BalletDuckNinja. I have recently acquired a NS and it is in 10.2.0 so it seems I have to use the latest version of Hetake.

I want to revert the lines of the nyx gui files from the commit you pointed out to see if it works.
I'm not familiar with the dev tools required, could you tell me where can I find more info?
I use Ubuntu and I've used ARM's cross-compilers in the past.

BalletDuckNinja, 2020-11-20:
> I'm having the same issue than BalletDuckNinja. I have recently acquired a NS and it is in 10.2.0 so it seems I have to use the latest version of Hetake.
>
> I want to revert the lines of the nyx gui [...]

As a workaround, you can use an earlier version of Nyx for now. Download the files for Hekate 5.2.1 and place them on your SD card, but inject the latest hekate payload instead.

chispas90, 2020-11-20:
The workaround works, thanks.

CTCaer, 2020-12-03:
Nyx provides an option to disable the bigger power save.

Try setting `newpowersave=1` to `newpowersave=0` in `nyx.ini` and reply back.

chispas90, 2020-12-04:
Working using `newpowersave=0`.  I also tested with `newpowersave=1` and it fails as expected.

Thanks

CTCaer, 2020-12-05:
Nice, this will mitigate the issue for you two for now.

What's interesting is that it only happens on 2 users.
@BalletDuckNinja has samsung ram chips (id: 00).

What ram chips you have @chispas90 ? (Check in hw & fuses in Console info. Left side has a `DRAM ID`)

chispas90, 2020-12-05:
It is `00: Samsung K4F6E304HB-MGCH 4GB`

CTCaer, 2020-12-05:
Thanks. BalletDuckNinja also did some ram freq switch tests and all of them hanged, no matter how much reasonable delay was inbetween.
For some reason your chips do not like going into 800MHz. And it seems that dram id type is irrelevant, because most have the same samsung chips.

So nyx will continue to mitigate that via the `newpowersave=0` option.
Thank you both for testing that.

closed 2020-12-05 by CTCaer