CTCaer/hekate #494: Linux Flashing not working

Octoham, 2020-10-02:
i get the full version of the latest (3.0.1) l4t ubuntu build. I already have the partition made with hekate and I flash linux using the **Flash Linux** feature to flash linux. It takes about one hour and thirty minutes to finish the process. When I want to select the emuMMC, hekate does not let me select my linux partition even though the partition has a size that isn't 0x00000000 and is named **Linux**

CTCaer, 2020-10-07:
3.0.1 does not have working releases.
As for selecting linux partition for emummc, that's not allowed.
It skips such partition for now. Make a new issue with a `Feature Request` if you want that.

closed 2020-10-07 by CTCaer

Octoham, 2020-10-07:
then how am i supposed to use the "flash linux" function. there should be a wiki in the repo that adresses all the features @CTCaer

CTCaer, 2020-10-07:
First you need to get a proper image.
Try again at switchroot.org. 3.2.0 is officially released now.

Octoham, 2020-10-07:
then what. its flashed using hekate on an other partition

CTCaer, 2020-10-07:
What what?
You boot it

Octoham, 2020-10-08:
im probably being very dumb but... how do i boot on an other partition?

CTCaer, 2020-10-09:
You should find a L4T Ubuntu entry in More cfg menu. 

You don't select a partition to boot, you choose the boot entry to launch.

Octoham, 2020-10-09:
ah ok