CTCaer/hekate #495: Hardware check  --improvement--😍

Borrajo, 2020-10-08:
The other day in our group, a participant had a bootloop on a vulnerable but not hacked console. After many tests, we used a different operating system, Lakka, and we saw that the problem was due to the fact that the wireless network card broke and that is why Horizon was doing the bootloop. I was thinking that Hekate might have an option to verify all the console hardware (as it does with the SD card)
for example: Wireless card, buttons, touch, SD, cartidge slot, audio, RAM, etc.

Once again thanks to Hekate we were able to find the error and help others.

Thank you!

CTCaer, 2020-10-08:
Well some of these could be checked via hekate. But others are too much work only to have a simple check.
Wifi for example needs initialization of PCI/PCIE hw and bus and also a proper stack to do PCI comms.
And even with that you will still need to initialize an interface to interact and check that the 80211 phy part works.
(Only asking the wifi card, if it's ok, is not enough)

To sum up, some extra testing tools will maybe added in the future. I will assess that request when I finish with some l4t/android stuff.

closed 2020-10-08 by CTCaer

Borrajo, 2020-10-08:
thanks for taking the time to answer me!
you are doing an amazing job!