CTCaer/hekate #497: Is there any way to turn the screen off when in Hekate?

safijari, 2020-10-10:
The reason I'm asking is that I don't want the fan on my switch running when I'm not using it (and running Android the fan always stays even, even if it's at a low speed). I can always reboot into Hekate though and the fan seems to not perpetually run in Hekate (like, it'll run while cooling down but eventually stop).

So my thinking being that if I can turn the screen off in Hekate then I can treat it as a sleep state for my setup. I can of course just turn the switch off but then I'd have to do the jig/payload business every time.

CTCaer, 2020-10-10:
The reason you hear the fan is because your android is Oreo which does not have a real sleep mode. It just idles with screen off.

You will need to upgrade your android build eventually.
Most sleep (real sleep) issues are fixed in android Q and as long as you remove the joycon it can sleep and wake without hanging.

hekate will eat your battery even if you disable screen, slowly but in a day, so it's not a good place to leave it for "sleep".

If switch had a notification led built in, I would have a screen off feature in hekate.
The only sensible thing for now, that would actually make sense in hekate, would be a dimming thing.

closed 2020-10-10 by CTCaer

safijari, 2020-10-11:
The battery bit I don't mind since it'd be plugged in anyway. A dimming function that dims super low would be good too

I'm on Pie right now. Built Q yesterday. Will try it.

Masamune3210, 2020-10-11:
Even if you leave it charging all the time, unless the switch completely disconnects the bat from the circuit, which I doubt it does, you would end up burning up your battery