CTCaer/hekate #501: Having Issues creating backup files.

Buwutiful, 2020-10-15:
Whenever I run the Backup eMMC BOOT0 & BOOT1 option, i keep getting the "File not found or could not be loaded!" and "Verification failed..." text. Unless there is a solution to this already, I'm not entirely aware of it yet. I'm using a Sandisk Ultra 256GB microsd XC1. That has already been Fat32 formatted.

Buwutiful, 2020-10-15:
A similar thing happens with using eMMC RAW GPP, but it goes on for an hour until it gives me an error.

Buwutiful, 2020-10-16:
I figured out I had a slight format issue so this is no longer a problem for me.

Buwutiful, 2020-10-16:
Nevermind, I thought it was fixed, but the issue popped back up again.

Buwutiful, 2020-10-17:
Alright I'll keep it closed... in the end the sd card was corrupted. :(

closed 2020-10-17 by Buwutiful

CTCaer, 2020-10-17:
Get a photo from Console info -> SD card