CTCaer/hekate #502: Waiting for LUN.

naldo29, 2020-10-17:
How do i fix this on UMS i cant use it, it says waiting for LUN and then it says timed out please help thanks!

CTCaer, 2020-10-28:
That's normally when host does not enumerate the usb device.
Check your cable or that your usb hub (internal-chip or external hub) can provide enough power.

naldo29, 2020-10-28:
> That's normally when host does not enumerate the usb device.
> Check your cable or that your usb hub (internal-chip or external hub) can provide enough power.

well my usb hub is connected to the wall, also i tried connecting it straight to my pc through usb 3.0 and it didnt work, do you think it might be a driver issue? maybe i messed that up?

naldo29, 2020-10-28:
also the usb gamepad works fine

CTCaer, 2020-10-28:
Well normally after the OS gets the device descriptors and sees that it's a UMS device, it sends a LUN request that tells the device what drive to mount. In this case it's always 0. As per specifications this must be sent before proceeding to enabling UMS.
What OS you use?

naldo29, 2020-10-28:
> Well normally after the OS gets the device descriptors and sees that it's a UMS device, it sends a LUN request that tells the device what drive to mount. In this case it's always 0. As per specifications this must be sent before proceeding to enabling UMS.
> 
> What OS you use?

Win 10

CTCaer, 2020-10-30:
Check if you use some weird usb filters or a custom mass storage driver.
For some reason it never sends the LUN request.

If you are up to the task you can also download wireshark, install it (select usbpcap) and then run it as admin and select your usb hub where switch is connected. Then inject hekate and start ums.

naldo29, 2020-10-31:
> Check if you use some weird usb filters or a custom mass storage driver.
>
> For some reason it never sends the LUN request.
>
>
>
> If you are up to the task you can also download wireshark, install it [...]

[image][image]thats the driver and im gonna try the wireshark thing although ive never tried it before.

CTCaer, 2020-10-31:
Yeah somehow you installed libusbK driver on the mass storage device.
Proper is:
[image]

Hit `update driver` and choose `USB Mass Storage Device` driver.

closed 2020-10-31 by CTCaer

naldo29, 2020-10-31:
Thanks so much!