CTCaer/hekate #503: New SD card and..., everything's broken now?

GamingInfinite, 2020-10-25:
Long story short I replaced my micro sd card, and now whenever I boot into hekate payload, even after I updated my bootloader folder, it comes up with LP0 library missing and Missing or old Minerva Library.  I don't know what broke, but I hope someone here can help fix.  Tried completely reformatting my micro sd to fix, but no dice.

impeeza, 2020-10-25:
Hi. For support you can read the documentation, go to discord or read tge wiki.
Saying that, if you're using hidden partition for EmuMMC the offset could change so you need to update the ini files to match.
Read the documentation about the archive bit too

urherenow, 2020-10-25:
Simply formatting isn't enough. New, Large cards tend to have 32MB unallocated space in the front. Get rid of that. Your main (1st) partition needs to be at the very start. Had that problem with a 3ds recently...

GamingInfinite, 2020-10-25:
@urherenow, I think that could be the issue, I'll try and report back.

GamingInfinite, 2020-10-25:
Nvm, I got it by downloading a seperate program, deleting all of the partition data then using the entire thing to create one singular partition

closed 2020-10-25 by GamingInfinite