CTCaer/hekate #506: Write Error   Com failure 3

fennectech, 2020-10-30:
I get alot of "write error" and "com failure 3" and the disk umounts when using USB mass storage

fennectech, 2020-10-30:
Turns out this is related to failed writes on a sdcard.   A corrupted file  being replaced from caja (forked version of gnome's file manager) causes the falure

CTCaer, 2020-10-31:
If there are failures to write SD, OS is notified and tries again and again. If it continues to fail it automatically rejects the whole drive. I can't do anything on that. As for com failure, it's a usb port/cable issue.

closed 2020-10-31 by CTCaer

fennectech, 2020-10-31:
Its definitly not a port / cable error. As the drive isn't being touched.  And it works fine if i copy the content to another directory on the drive and cut and paste it to the right locations.  But i see the issue with filesystem corruption.    The port and cable are fine and haven't had issues for anything else.  Also. Using cp from the command line to copy the data works just fine.   It usually applies to 0 byte files that are created by a failed write.

NotPikaQuin, 2025-06-08:
> If there are failures to write SD, OS is notified and tries again and again. If it continues to fail it automatically rejects the whole drive. I can't do anything on that. As for com failure, it's a usb port/cable issue.

it is a usb port error thank you