CTCaer/hekate #511: [ERROR] : NXBoot failed to mount the EmuMC.

Deadfalt999, 2020-11-21:
This error happens randomly for me, sometimes while I transfer files from my computer to the sd card and the put the sd card back in the switch it happens. I just don't get it, I didn't touch anything in particular. Any insight as to what could be doing this would be nice because i'm completely lost. I have made backup of my sd card and generally when I restore those backups img it works again and I can access it.

Deadfalt999, 2020-11-21:
I made my emunand with hekate to be more precise and don't know where to go to find help anymore, if this isn't the place i'm sorry but any sort of insight would be helpful.

CTCaer, 2020-11-22:
That error comes from fusee primary and not hekate.
On the other hand your sd card is probably fake.

Got to Nyx -> Tools -> sd card and get a photo or screenshot and paste it here

Deadfalt999, 2020-11-22:
I've verified my sd card with h2test though. Everytime it appeared as a real sd card without any issues. You want a screenshot of the micro sd card info ?

Deadfalt999, 2020-11-24:
Alright I solved my issue by using nx nand manager instead of using the other backup method I was using.  Sorry to have bothered you and have a good day.

closed 2020-11-24 by Deadfalt999