CTCaer/hekate #512: Sd and Emmc data <@LBA 033B0000> do not match

Tayden0113, 2020-11-27:
I started to perform a full GPP Nand backup, and it got about 88% through the backup when it gave me this error. I looked up the issue and it seems this has happened before but with different codes. Should I downgrade to an older version of Hekate? Is there a step I'm missing or something I didn't do correctly? Here's the guide I used. https://nh-server.github.io/switch-guide/user_guide/emummc/sd_preparation/

impeeza, 2020-11-28:
Did you test your SD card, for me the erorr was a fake SD.

Tayden0113, 2020-11-28:
> Did you test your SD card, for me the erorr was a fake SD.

How would I do that?

impeeza, 2020-11-28:
You can use h2testw to test the sd on the pc,

CTCaer, 2020-12-02:
That's because your sd card is probably fake. Post a screenshot or photo of Tools -> SD card info from Nyx

CTCaer, 2020-12-09:
Based on lack of reply from user, seems that the sd card was changed to a proper one.

closed 2020-12-09 by CTCaer