CTCaer/hekate #513: UMS consistently phase-errors

rico-387, 2020-11-29:
This is the closest to a place for a bug report or a support ticket that I could find. I'm trying to transfer data from my SD card to my PC via hekate's built in UMS tool. About 15 seconds in, it stops and gives status: phase-error for a few seconds before ejecting itself, making data transfer a mildly frustrating process. I'm fairly sure this isn't a hardware issue, as this didn't happen before I updated hekate and atmosphere, and I've tried with several USB cables and configurations.

masagrator, 2020-11-29:
Atmosphere version doesn't matter. So go back to hejste version that worked and check again for errors

CTCaer, 2020-12-02:
Phase errors are normally because of problematic usb port or plug or cable.

CTCaer, 2021-01-05:
@rico-387 Test with the latest version. It has more relaxed timings.

closed 2021-01-05 by CTCaer